360Controller/360Controller #493: Microsoft Wired Xbox 360 Controller shows up as two controller in Steam

david-cao, 2017-07-06:
When using a Microsoft wired controller on Steam, two controllers seem to appear. All buttons except for the start button act only on the "Xbox 360 Controller". When pressing the start button, it seems to trigger both controllers. Specifically, when playing Rocket League, the game both paused and added a new player. Both players were able to be controlled simultaneously. 

[image]

fuzzy76, 2017-07-29:
Seeing the same with a Xbox One controller. Portal 2 behaves weird because of this (any analog stick controls both movement and camera).

FranticRain, 2018-04-03:
Steam contains native support for Xbox One controllers. So what you're seeing is to be expected. You have a driver that is advertising the device, and an application that is bypassing the driver and adding a device. You can disable to the driver from the preference pane when playing games where this is an issue.

The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

closed 2018-04-03 by FranticRain