360Controller/360Controller #494: Conflicts with Steam Controller

scuddermj, 2017-07-07:
When 360 Controller Driver is installed, Steam does not recognize the Steam Controller.  It is recognized as a 360 controller instead, and refuses to pair.

closed 2017-07-07 by scuddermj

jgonfer, 2017-07-07:
Has Steam recognized your Steam Controller? I'm interested on this because this week I bought a Steam Controller with the Summer Sales 🤔

scuddermj, 2017-07-08:
My mistake actually.  It had been a problem in the past.  I decided to check for new versions of the driver this evening, saw nothing that addressed the issue, and I still couldn't pair.  Steam had an update queued, and after applying it, the controller paired just fine.  I'm on the Steam Beta track, if that makes a difference.

jgonfer, 2017-07-08:
Thanks for the info @scuddermj, good to know 👍