360Controller/360Controller #495: 360 receiver not working?!

varoudis, 2017-07-10:
I got this generic receiver that I tested it in a fresh Win10 (for sanity) and it did worked, linked with the 360 controller and 'activated' player 1.

Latest OSX gives:
Product ID:	0x02a9
Vendor ID:	0x045e  (Microsoft Corporation)
Version:	6.0a
Speed:	Up to 12 Mb/sec
Location ID:	0x14300000 / 5
Current Available (mA):	500
Extra Operating Current (mA):	0

But controller wont connect. 
Any ideas? What sort of dmsgs should I look for?

poorlymac, 2017-08-11:
Hi, I figured out how to make this work and am working on a patch for this, and having a go at adding the 360 Racing Wheel (basing on what I see in the One Racing wheel and Linux xpad). the temporary hack trick is to edit /Library/Extensions/WirelessGamingReceiver.kext/Contents/Info.plist and change the idProduct value in the WirelessGamingReceiverForWindowsAlternate section from 657 (i.e. 0x0291)  to our units value 681 (i.e. 0x02a9) then kextunload and kextload (or reboot), you may also need disable SIP. I did also alter code in 360Daemon but I'm not sure that was required in the end but will figure it out at some point. Tell me how you go!

varoudis, 2017-08-13:
it;ll take me some time to test as Im on transit. Ill let you know though.
Thanks

nour-s, 2017-11-09:
@poorlymac That's awesome, your solution worked like a charm.
This answer should be referred to all the opened issues that have the same problem.

By the way, I tried to leave the old value as a comment, 
``
but it didn't work until I removed the comment, also I had to run 
`sudo kextload /Library/Extensions/WirelessGamingReceiver.kext`
first, in order for the unload to work:
`sudo kextunload /Library/Extensions/WirelessGamingReceiver.kext`
and then do the load one more time!

Endogen, 2018-02-26:
Is it possible to merge this? Or why isn't it merged already?

ohiocitydave, 2018-07-31:
I followed poorlymac's advice and it worked like I charm; pesky file permission adjustments notwithstanding. I did not have to disable SIP and did not change any code in 360Daemon.