360Controller/360Controller #498: Rumble options?

Mjhietpas, 2017-07-15:
I'm using an xbox360 controller and I can't seem to turn off rumble when I pull the triggers. Is there a way to do this?

shangles, 2018-01-31:
I would also like to know this.  I'm using an Xbox 360 wired controller on a MacBook Air running 10.13.3.

When playing Rocket League whenever I press the trigger to accelerate in the game it rumbles.  I'm forced to disable vibrate in the System Preference.  It's an unfortunate workaround as I prefer having rumble enabled in-game.

electric-monk, 2018-01-31:
Are you leaving the Preferences open while playing the game? As a test it will operate the rumble motor corresponding to the trigger, but only while Preferences is open.

shangles, 2018-02-05:
Well, no matter what I try, I can't get rumble to work in game with Rocket League for Mac (Steam)  I have controller vibration enabled in game.  I've tried toggling the driver off/on in the System Preferences.  Verified it's enabled there (Default) but no vibration at all for any events in-game.   This is using a wired xbox 360 controller.

FranticRain, 2018-04-03:
The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

closed 2018-04-03 by FranticRain