360Controller/360Controller #499: Third party wired xbox360 controller is unrecognised

lera2, 2017-07-17:
''No device found''

Mac El Captain 10.11.6 

Thanks

Generic   USB  Joystick  :

  Product ID:	0x0006
  Vendor ID:	0x0079
  Version:	1.07
  Speed:	Up to 1.5 Mb/sec
  Manufacturer:	DragonRise Inc.  
  Location ID:	0x14200000 / 10
  Current Available (mA):	1000
  Current Required (mA):	500
  Extra Operating Current (mA):	0

343678573, 2017-11-02:
me too , i can't find any way to solve the problem ,if u find some way ,pls tell me ,pls

soswow, 2017-12-26:
Same cheap Chinese clone here. Well, it seems like I will have to spend more money on a proper controller =\

soswow, 2017-12-26:
I think this one is a duplicate of https://github.com/360Controller/360Controller/issues/299

FranticRain, 2018-04-03:
As noted in #299 this controller is not supported.

closed 2018-04-03 by FranticRain