360Controller/360Controller #503: Binding Options (RT - LT)

PaulAChambers, 2017-07-22:
After it all installed, i mapped my xbox one controller but then realised that RT and LT aren't an option, i went onto my chosen game and realised they weren't working and wondering if their was a way i could get them to work

Rogue-Hosaka, 2017-09-26:
Hello, could you solve this problem?

FranticRain, 2018-04-03:
The triggers are analog inputs, not digital. They cannot be remapped using the binding tool because of this.

The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

closed 2018-04-03 by FranticRain