360Controller/360Controller #504: Any updates on Sierra 10.12.6 wireless compatibility

bfeinstein, 2017-07-25:
I am unable to get the driver to recognize my wireless 360 controller. Any advice for me?

PKBeam, 2017-07-31:
I'm having issues with a wireless 360 controller, on High Sierra. My friend running Yosemite doesn't have these issues.

FranticRain, 2018-04-03:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#wired-xbox-360-controllers

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain