360Controller/360Controller #505: PDP xbox one controller not working

mollywhopped, 2017-07-26:
So I have tried to download every last driver and nothing will work, I downloaded 360Controller 0.16.5 and what do you know it.. it worked! I played rocket league for about an hour and logged off. When I came back (about 2 hours later) it wasn't working again. What I mean by not working is the driver said I had a controller connected but no buttons were working. I have OS Sierra 10.12.3    I was just thinking about returning the controller and getting a new one. The controller that I want to get is made by "Power A" will this controller be compatible with the current drivers that are out?

nightdr3am, 2017-09-05:
nope.....

lynchy901, 2017-09-20:
https://github.com/360Controller/360Controller/issues/469 Try this.

closed 2018-04-03 by FranticRain