360Controller/360Controller #507: XBOX ONE CONTROLLER RIGHT STICK NOT WORKING.

DeltaSqwad, 2017-07-30:
When i connect my xbox controller it works pretty fine in the driver and it shows the right stick is moving probably...BUT in SOME games, not all Of them the right stick just STOPS WORKING...simply doesn't move or recognized, please help.

FranticRain, 2018-04-03:
The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

closed 2018-04-03 by FranticRain

DeltaSqwad, 2018-04-03:
I understand, and you're right its definitely the games job to support controllers, but you gave achieved something the whole system denied, that's why I asked.
But I understand and I really do appreciate your work, thanks alot.