360Controller/360Controller #509: Controller not working

oli1220, 2017-08-12:
So when I go into the driver the controller is detected but wont respond to any of my inputs of the controller and in games it won't listen to any of my inputs either.
[image]

Help please
Mac Version 10.12.6
13 Inch Macbook air

Controller
Power A Spectra Illuminated Controller

wmaness, 2017-09-04:
I have this  problem as well, Power A, wired USB controller:

  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro11,3
  Processor Name:	Intel Core i7
  System Version:	OS X 10.11.6 (15G1611)
  Kernel Version:	Darwin 15.6.0
  Boot Volume:	Macintosh HD
  Boot Mode:	        Normal

pulkitag, 2017-09-27:
I have the same issue, with the same controller. Any suggestions on what to do? @oli1220 did you find a solution?

PokeCaptain, 2017-10-03:
What works for me is to plug in the controller, then disable and reenable the driver. Seems to work, but it's a bad solution long-term.

closed 2018-04-03 by FranticRain