360Controller/360Controller #513: Preferences not saving

michet55, 2017-08-20:
I'm trying to use the spoof Pretend to be an Xbox Controller, but every time I check the box and close system preferences, it unchecks itself. The same if I try to make any other changes in system preferences for it. Is there an apply changes button I'm missing somewhere?

zapstar, 2017-08-30:
I'm assuming that the developers are choosing not to persist the "Xbox360" spoofing option. Let's see what they say.

Ratamorph, 2017-10-27:
I have the same issue, it started happening in High Sierra for me.

Niblitlvl50, 2017-12-21:
Same here. This is especially annoying when developing against SDL and their controller api does not recognise the controller unless i use the spoofing option. [macOS Sierra, 10.12.6]

FranticRain, 2018-04-03:
This build contains a fix for preference save issues. Could you check it out and tell me if it works? 
[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/1869822/360ControllerInstall.dmg.zip)

Niblitlvl50, 2018-04-15:
Does not work for me. The version that was installed by that archive is: "Version 0.16.5". I check the "Spoof" checkbox, close system preferences, goes back in again and its not checked. The controller works though, until i restart my computer or unplugs the controller from the usb port.

FranticRain, 2018-04-16:
Then this isn't fixed by the fix proposed in https://github.com/360Controller/360Controller/issues/451#issuecomment-340976068

FranticRain, 2018-06-01:
Settings bugs should be fixed in the latest release.

closed 2018-06-01 by FranticRain