360Controller/360Controller #514: Request: PowerA/Liquid Metal Controllers

edwin-pers, 2017-08-22:
Hi,

Some 3rd-party PowerA/Liquid Metal xbox 1 controllers need to be sent a rumble packet at init before they'll send input. 

See here:
[https://github.com/paroj/xpad/issues/48#issuecomment-284171968](https://github.com/paroj/xpad/issues/48#issuecomment-284171968)

and here:
[https://github.com/cgutman/xpad/blob/for_upstream/xpad.c#L380](https://github.com/cgutman/xpad/blob/for_upstream/xpad.c#L380)

for what had to be done to get these to work on linux.

Any way you can implement that in this driver?

Thanks!

nightdr3am, 2017-09-05:
hi i have a PowerA/Liquid Metal Controller. But i was wondering if theres any development toward this i would like to use my third party xbox controller. i am struggling to to get it run on the latest Mac OS at the current time.

blakekl, 2017-11-20:
I'd also like to see this. Newegg just had a sale for these things at $10 a pop, so I bought a few. They work fine on my windows box, but not on my linux/mac boxes.

FranticRain, 2018-04-03:
Here's a build with the proposed fix. Let me know if it works. 
[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/1869818/360ControllerInstall.dmg.zip)

blakekl, 2018-04-03:
I just tried the proposed fix. There is no difference in behavior. The controller shows up in the dropdown list as Xbox One Wired Controller (Wired), and the light on the controller is on, but none of the buttons or axes show any input when used.

blakekl, 2018-04-03:
For what it's worth, I cannot get these to work on linux either. They work just fine on my windows machine though.

FranticRain, 2018-04-03:
Okay, then the assumption that this is an IOUSBHost issue holds. I pushed the fix anyway. Closing this.

closed 2018-04-03 by FranticRain