360Controller/360Controller #525: High Sierra incompatibility? (system wouldn't boot)

kdorff, 2017-09-29:
I upgraded my 27" iMac 5k to High Sierra. Before upgrading, the machine booted fine (and was running the latest patches of Sierra). Afterward, it wouldn't boot. The progress bar would fill up but never complete. Could boot into safe mode. First Utility First Aid on my disk showed no issues. 

I turned on Verbose during boot. The messages before I gave up look like
https://photos.app.goo.gl/gNtx2z5EugTmPbok1

I removed WirelessGamingReceiver.kext, Wireless360Controller.kext, and 360Controller.kext and then my system booted without issue.

pxia, 2017-09-30:
I have problem with 10.13 as well. My system booted well but the controller doesn't seem to work.

Trevor-Z, 2017-10-01:
Yeah, same here. The latest release of the driver isn't working on High Sierra, the controller isn't detected. But I don't have any issues with booting.

PokeCaptain, 2017-10-03:
Interesting. I've had this driver for a couple of years now, and the upgrade to High Sierra does very little. Once in a while, the OS notifies me that the driver is incompatible, but it's sporadic. As far as I'm aware though, It's never caused a boot issue. Interestingly enough, the most reliable way for me to get a controller working is to have the controller plugged in while it's booting. Plugging it in afterwards requires that the driver be disabled and then reenabled to work properly.

owenzhao, 2017-10-03:
@pxia @Trevor-Z You need to keep you Xbox One S controller power on when you install this app.

1. wire your controller with usb dongle
2. press the Xbox button, so it slowly blinking. Be sure not pressing the bluetooth connecting button.
3. install this app again. if the controller stop blinking, press the xbox button again.
4. reboot.
5. when enter the system. Goto `system settings` -> `privacy setting` -> allow `Drew Mills` app. You controller is not recognized because macOS High Sierra prevents the driver.
6. You controller will stop blinking and the light keeps on. You can use your controller now.

If you were on macOS 10.12 or lower, the system won't stop the driver. So you wouldn't have any issue. Just like @PokeCaptain , I thought he must upgrading 10.13 from 10.12, so the driver is not prevented.

Trevor-Z, 2017-10-04:
It's a wired 360 controller. I just tried it now, and it worked.

I had allowed the "Andrew Mills" app yesterday, thinking it was related to some other thing.

Thanks!

pxia, 2017-10-07:
@owenzhao I can't find the `Drew Mills` app. Is this the screen what you are talking about?

[image]

owenzhao, 2017-10-08:
@pxia It only shows the first time you reboot from installing this app. And it is in `General` tab, not in `Privacy` tab.

You need to reinstall xbox 360 controllers and reboot your mac to see it.

Here is my screenshot in Chinese.

[image]

kargq, 2017-10-14:
yeah it shows its connected in system report but dosent seem to work.

owenzhao, 2017-10-14:
@kargq For Xbox On S controller, it always shows in USB section in system report and can be used in emulators like OpenEmu. But you have to make it show in 360 controllers (this app) and enable setting fake to be a Xbox 360 controller to let it work in Stream downloaded games.

Kraci, 2017-10-27:
@owenzhao Thank you so much, Drew Mills was the thing :) 👍

addzo, 2017-12-26:
Still have this issue (or similar?) after following the suggestions on this page.
In my case, 
I can see the controller in the "Xbox 360 controllers" section of the System pReferences
"Drew Mills" never shows up as an untrusted driver even after uninstalling and reinstalling
In the systems settings app I can watch all the components move while I move the controller but not a single game I have recognizes the controller.

Any ideas?  Thanks for the help.

chrismower, 2018-01-03:
Followed @owenzhao 's instructions and worked perfectly. Thank you!

A few helpful points of clarification:

1) The exact location for driver approval is _System Settings > Security & Privacy > General tab_. (See @owenzhao's second screenshot.) This will only display after the driver install reboots your system to finish. 

2) If your controller is now working via cable, but not wireless, you may have to disconnect your other bluetooth devices first. For some reason, my mac (running 10.13.2) could not find the Xbox controller until I had disconnected my bluetooth headphones. When I did, the controller then popped up as an option to pair. Once paired with the controller, I could reconnect my bluetooth headphones.

3) A direct link to the driver download is here, so you don't have to dig through the repository to find it: [360 Controller Driver Releases](https://github.com/360Controller/360Controller/releases)

After following the provided steps, my device is working with Steam games.

glemiere, 2018-03-05:
@addzo In the preferences, go to Xbox 360 Controllers --> Binding --> Reset Mapping.

FranticRain, 2018-04-02:
To respond to the original issue, there is a known issue with 360 wireless adapters in 10.11+. Further discussion of this issue has been moved to #605

closed 2018-04-02 by FranticRain