360Controller/360Controller #527: Can we please have a signed build of the bluetooth rumble merged pull request?

MichaelMKenny, 2017-09-30:
The title says it all. I was really happy to find out today that MaddTheSane got this working! I spent the afternoon trying to compile and code sign it myself, but I'm on sierra and Xcode 6.4 doesn't play nice with it. I did manage to hack Xcode 6.4 to run but ran in to issues code signing the kexts properly.

It would really mean a lot to me if you could upload a signed build with these changes. I'm really eager to test it for myself :) Thx

matthewdias, 2018-01-26:
@MaddTheSane or anyone else who has access, would you be able to create a new release? I also tried to build and got as far as
```
ERROR:

Description:

Unable to find signing certificate
==============================================================================
Build Failed
```
I would try it in a vm, but i dont have any storage space left.

FranticRain, 2018-04-02:
Does this build include the stuff you're looking for? [360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/1869747/360ControllerInstall.dmg.zip)

MichaelMKenny, 2018-04-03:
Thanks. Just tried it, it does seem to include the XboxOneBluetooth.kext, but the controller disconnects quite often, and it doesn't rumble. From @MaddTheSane's commits I wasn't sure whether he'd actually got rumble over bluetooth working, it was a little unclear. 
By the way I'm testing rumble with Steam In-Home Streaming, as I don't play games natively on my mac. Rumble does work when playing over Steam In-Home Streaming using the normal 0.16.5 build with my Xbox One controller connected with cable.

Also, installing this build seems to have stop the Xbox One Controller from rumbling when connected via micro usb. Anyway thanks for building it for me, I appreciate it. But, I will be going back to 0.16.5.

FranticRain, 2018-04-03:
Okay. Going to close this, since the build has been provided.

closed 2018-04-03 by FranticRain