360Controller/360Controller #529: Xbox 360 Controller not recognized by driver

hferrone, 2017-10-02:
Hello,

I thought I bought an Xbox 360 controller, not a third party, but I'm not sure. Here is a screenshot of my device ID etc. Can someone please let me know why the driver isn't recognizing the controller?

Cheers!
[image]

hferrone, 2017-10-02:
System specs
[image]

PokeCaptain, 2017-10-03:
Performance Designed Products is third party. Seems to be one of [these](http://pdp.com/en/shop/xbox_360/).

kargq, 2017-10-12:
same prob, and my controller is totally originaly brand new x box one s controller
[image]

Carsonsmuts, 2017-11-07:
Ditto

[image]

FranticRain, 2018-04-02:
PDP and PowerA Xbox One controllers are incompatible with macOS 10.11+ thanks to a rewrite of the macOS USB kernel. We cannot resolve this issue. It is a bug in Apple's core OS code.

Please open new issues for different controllers.

closed 2018-04-02 by FranticRain