360Controller/360Controller #531: Xbox One controller won't respond.

MegaMug, 2017-10-11:
I got a Xbox One controller (Most likely 3rd party. Says PowerA on the back. :P) the other day so that I could use it to play Cuphead. My Mac recognizes that the controller is there, but when I'm at the test screen, nothing on the controller responds. I know it works because I tried it on a windows computer and all the buttons worked. My current OS is Sierra version 10.12.6 if this helps. I appreciate any and all assistance.

PokeCaptain, 2017-10-11:
This seems to be a long existing bug that I have as well. As a workaround I use, you could disable and reenable the driver while the controller is plugged in. Alternately, you could start up the computer with the controller already plugged in.

MegaMug, 2017-10-11:
Okay, just so I know and don't screw it up, how would I disable and re-enable a driver?

PokeCaptain, 2017-10-11:
I just tested mine, and it seemed to work without going through this process. However, I recorded it anyways.

https://drive.google.com/file/d/0B0EedKo8lurWaV83YVAwU3hmcHM/view?usp=sharing

MegaMug, 2017-10-11:
I tried a restart, shut down, and the enable and disable. Nothing. However, it knows it's there because it says "Xbox One Wired Controller (Wired)",

MegaMug, 2017-10-12:
Just to update on some info, here are the specifications of the controller.

Xbox ONE liquid metal controller:

  Product ID:	0x543a
  Vendor ID:	0x24c6
  Version:	1.01
  Serial Number:	0000000000000000
  Speed:	Up to 12 Mb/sec
  Manufacturer:	PowerA
  Location ID:	0x14200000 / 13
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

godber, 2017-11-11:
I have the same device with the same result (no input).

The video provided by @PokeCaptain was useful since it showed me what the Settings pane should do for a working controller.

I tried disabling and enabling the driver as shown in the video but there is still no input.  I tried restarting with the controller attached.  I tried these both on macOS 10.13 and then upgraded to 10.13.1.

tommyjackson, 2017-11-17:
I'm having the same problem. Literally just went and bought this controller only to come back and have it not respond. Like the others it shows up in the preferences, but doesn't seem to register any buttons at all.

Melechim, 2017-11-19:
Ok so I have one of the launch controllers and I also have the Power A cable neither will work on my 2017 MacBook Pro I need some insight. Just trying to play Fortnite.

Melechim, 2017-11-19:
It won't turn either of them on my original controller at least rumbles when I plug it in I also do have a data cable.

apb514, 2018-03-13:
I have the same Product ID but the system information recognizes it as "Xbox Pro Ex controller" with Product ID: 0x543a
Vendor ID: 0x24c6

Controller won't power on (no green light) and it does not get recognized in preferences.

FranticRain, 2018-04-02:
PowerA Xbox One controllers are incompatible with macOS 10.11+ thanks to a rewrite of the macOS USB kernel. We cannot resolve this issue. It is a bug in Apple's core OS code.

closed 2018-04-02 by FranticRain