360Controller/360Controller #532: HORIPAD Xbox One controller not detected..

dreaduu, 2017-10-13:
Hello everyone.

Im trying to use HORIPAD controller for xbox one on my Macbook Pro.
- [https://www.windowscentral.com/hori-pad-pro-review](url)

Unfortunately the controller is not deteted. It does light up when I connect it before turning on the macbook but nothing happens after that and if I disconnect it - it doesnt light up anymore.

Any suggestions? Im staying at my girlfriends house and we really want to play Death Road to canada which is a great game (try it!)

FranticRain, 2018-04-02:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

HanSeoulOh, 2018-04-08:
I have this controller too..

Details that system report shows:

HORIPAD ONE:

  Product ID:	0x0100
  Vendor ID:	0x0f0d  (HORI CO., LTD.)
  Version:	1.01
  Serial Number:	0000C321D742219F
  Speed:	Up to 12 Mb/sec
  Manufacturer:	HORI CO.,LTD.
  Location ID:	0x14300000 / 7
  Current Available (mA):	500
  Extra Operating Current (mA):	0

closed 2018-04-08 by FranticRain

FranticRain, 2018-05-17:
This device is in the latest build.

commit e3dd9c30 (2018-04-08): Add device from #532

Resolves #532