360Controller/360Controller #534: Steel Battalion controller compatibility?

mcamis, 2017-10-15:
Has anyone else got this working? I'm seeing regular OG Xbox controllers showing up in the device dropdown without issue, but not the Steel Battalion controller.

FranticRain, 2018-04-02:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#wired-xbox-360-controllers

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain