360Controller/360Controller #537: Xbox One Bluetooth Controller

Muelltonnen, 2017-10-18:
Hey there! I have a problem with the above mentioned controller.
It's being recognized by my mac and by steam. But i can't use it in Borderlands 2.

Would it be possible to connect it with an usb cable (since it's only being recognized by your driver with usb) and then to trick borderlands with the Xbox 360 trick option? Or is there a possibility that you do your driver also for the bluetooth controller? Or just the do a driver for the trick option? I would pay for that!

MaddTheSane, 2017-10-18:
Try it out on other games: some games may not recognize the Bluetooth controller as an Xbox Bluetooth controller. Your best bet is to contact the game developer and ask them to fix it, or use something that can emulate controllers.

Muelltonnen, 2017-10-18:
Thank you! But is it not possible for you to implement the bluetooth controller in the driver or do a driver with only the function to trick the game into thinking it's an 360 controller? :/

Muelltonnen, 2017-10-18:
Or your mean something like this: https://www.x360ce.com ?

And I've read that the USB Xone controller is supported by your driver so maybe a usb cable could be the solution?

Muelltonnen, 2017-10-18:
So the wired Xbox One Controller is working with the driver and the function to pretend to be an 360 Controller.

So I tried to connect my bluetooth controller with an usb cable, but the controller won't light up and won't show up in the driver settings. Is it a special cable I have to use that the driver will recognize my controller as wired?

Thanks in advance!

Muelltonnen, 2017-10-19:
Nothing?

closed 2017-10-19 by Muelltonnen