360Controller/360Controller #539: Official Xbox 360 controller not working with Steam

MegaMug, 2017-10-21:
As stated in the title, it won't work with steam. Is there anything that I can do to fix this?

elliotgeen, 2017-10-23:
I have got the same problem, my official xbox 360 controller has stopped working, the binding in the program is fine (Controller Test), its just when I go to steam or play for example rocket league it doesnt work (B is Y or spacebar). I think theres an issue with the steam reading the controller as keyboard buttons and not as controller buttons but i wouldnt know. Reason I am commenting is just now I have been trying to use my controller and it has stopped, searched on here and found this guy has the same problem as me. PLEASE HELP!! Thanks!

InterStena, 2017-10-28:
```
We cannot fix game specific issues. This driver does its absolute best to put out a standardized format for games to use. If they don't take advantage of that, there is ABSOLUTELY NOTHING we can do. The best we can do for you is give you the "Pretend to be an Xbox 360 Controller" option in the "Advanced" tab.
```
Try this.

FranticRain, 2018-04-02:
The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

closed 2018-04-02 by FranticRain