360Controller/360Controller #541: Wireless original XBox 360 not recognized

janisc, 2017-10-26:
I am running High Sierra 10.13 and 360Controller version 0.16.5.

Driver control panel says "No devices detected". I have a receiver and a controller that is linked to receiver connected.
ProductID:	0x0719
VendorID:	0x045e  (Microsoft Corporation)

This is original XBox wireless controller, for all I know.

dsc, 2017-11-01:
Identical setup and issue. In case it helps: pressing the connect button on the receiver and then the same on the controller *does* do the LED dance that indicates both went into pairing mode, and then found a mate. However, the controller does not show up in the driver's control panel / do anything.

frenangomes, 2017-11-05:
This issue helped me https://github.com/360Controller/360Controller/issues/525.

FranticRain, 2018-04-03:
The Xbox 360 wireless controller is not supported on 10.11 thanks to a rewrite of the macOS USB kernel. This causes kernel panics and unexpected issues with your mac.

closed 2018-04-03 by FranticRain

dsc, 2018-04-21:
Not disputing the facts of the matter, but this is really bad. The 360 controller is the *only* quality wireless controller that has broad game support. I bought a current-generation XB1 controller after carefully researching mac compatibility, steam compatibility, quality of play over bluetooth, the featureset, alternatives... Only to find out empirically that most the games I want it for have explicitly added 360 controller support, and generic controllers are spotty at best (but usually just inoperable). This includes the almost everything I tried in my library -- from Borderlands 2 to indie platformers, the 360 worked (now that it has paired), and the XB1 did not. Forum threads suggest the same for the DualShock, though I don't have one. Is there a plan, or do we just give up and (ugh) build a beige box?

FranticRain, 2018-04-23:
There isn't a plan, because there is literally nothing we can do. Apple changed the way drivers work, and it breaks this version of the driver. A new driver would require a modern rewrite of the core code. This would break compatibility with a bunch of games. This is because controller support on MacOS is hot ass, and the original version of this driver for 360 controllers more or less created a base standard for controllers on the operating system. So everyone and their mother that wanted controller support in their mac game came up with some weird way of doing it. For example, games like Awesomenauts just straight up grab the packet and interpret it as a 360 controller, and ignore the USB HID specification. That's why the "pretend to be a 360 controller" feature exists. Some games (I think all Aspyr ports, like Borderlands 2) have their own driver-like system of connecting to controllers built in to the OS in a way that actually isn't compatible with the driver. Having the driver turned on breaks the controller in Borderlands 2. It's a fuckin' DISASTER. The best we can do for the future of the controllers on macOS right now is to cut this off here with the advent of modern Bluetooth Xbox One controllers and tell everyone to get on board with standardized Bluetooth HID or use a legitimately good controller library like [DDHidLib](https://github.com/Daij-Djan/DDHidLib). I know this isn't what you want to hear, but it's all a disaster created by Apple, and then perpetuated with lazy and/or sloppy macOS ports.

wd30, 2018-04-24:
Honestly, "buy new stuff" is terrible advice. I would assume if many of us could reasonably go that route, we'd be building gaming PCs as opposed to trying to figure this out through other means. getting the wireless adapter to work is the only way to game for me at this point, and to hear that i have to suck it up until I have a few extra hundred bucks lying around (fat chance of that anytime soon) is terrible news. Now I can't code for shit, but I'd love to help out in any way that I can. 

To be clear, I know this is something this group is doing for nothing, and are only doing it out of the goodness of their hearts. What your doing is really awesome, and you've been doing it for a really long time. At this point, I wouldn't be able to game without you. I'm sorry Apple fucked up a lot of your hard work, but there has to be a better solution than telling a bunch of people to pound sand.

FranticRain, 2018-04-24:
I'm sorry, I must have come across wrong. I didn't mean to say that the solution for you is to buy something new. I'm simply saying that the best way to unify the controller situation on macOS is to unify around a standard built in to the operating system. The potential code-only fix to this problem would be a rework of the driver that matches the new IOUSBHost standard. These standards are cryptic and poorly documented because Apple doesn't want kexts to exist. But this is the format in which all kexts past 10.11 are supposed to be in. The old style of kext is supported through a broken compatibility layer.

Now, is that possible? Yes. Certainly. Now, the more pressing question is whether or not it's actually a legitimate fix. Like I said before, every game has it's own quirky way of interacting with the controller. It isn't clear whether or not an IOUSBHost based driver would even work. We can take look back at my personal project Xone-OSX for an example of how quirky this is. The original iteration of Xone-OSX was built practically identically to this driver. Heck, it was a straight rip off but with hooks for Xbox One controllers instead of Xbox 360 controllers. But, it didn't work with many games because games were hard coded for the TattieBogle driver. Now, the wireless controller work is an different beast entirely from the wired work that I did, and far more complex. So your guess is honestly as good as mine on whether or not it'll actually fix the issue.

So, what's the low cost fix? It isn't pound sound. It's downgrade to 10.10. Plain and simple. I know. It isn't exactly an exciting or optimal fix. But it works. Right now.

dsc, 2018-04-24:
Thanks for the additional detail. And +1 to the thank-you-for-thankless-work.

FranticRain, 2018-04-24:
We can spoof a wired Xbox One controller as a wired Xbox 360 Controller. The Xbox 360 Wireless controller is an entirely separate kext, so there isn't any process to accomplish that. We could give the Xbox One controller the same VID/PID as a wireless receiver, but that wouldn't likely fix any issues, since programs that are operating on that level of granularity are going to be looking for the UID of the wireless driver, rather than the wired one.