360Controller/360Controller #543: Logitech Chillstream (similar to xbox 360 gp) not recognising in Parallels Desktop

InterStena, 2017-10-28:
I have a windows 7 installed on my parallels desktop VM, I play a lot of games with it.

Controller exists in Mac (works in Super Meat Boy when "Pretend to be xbox 360 controller" enabled). But absolutely doesn't work in virtual machine. Logitech says it should work in windows without any drivers.
It sounds more like question but idk where to ask it. So hope developers/community/anybody will help me to fix this.

InterStena, 2017-10-28:
(some info about parallels)
All my usb devices should appear in this section:
[image]
But I don't see my controller there.

FranticRain, 2018-04-02:
This controller is not supported by official Xbox hardware. As such, it's support is outside the scope of this driver. We cannot and will not support this driver.

closed 2018-04-02 by FranticRain