360Controller/360Controller #545: No devices found

danvtim, 2017-10-31:
I am using an circa 2011 iMac with high Sierra. 

I can attach and see three different wired controllers all of which are defined in the packages info.plost.

So, everything should just work no? 

But, I just see no device found. 

Any help would be great. Currently using the Rock Candy controller.

danvtim, 2017-10-31:
IT turns out I had to enable permission for the extension to run the the security prefs. Now, it's working.

343678573, 2017-11-02:
hello ，could u tell me how to enable permission for the extension to run the the security prefs. my english is not very well. i don't know could u understand what i say. And i just use the translation tool to catch on your mean

nour-s, 2017-11-09:
I have the same problem, I'm using wireless receiver to connect XBOX 360 controller.
The controller seems to connect to the receiver (it stops blinking when I press the button, but it still flashes slowly), but the software is still showing  (No device found).

dmcb, 2017-11-10:
Same exact problem here. Drivers worked for me forever, but on a recent fresh install of High Sierra, I cannot get my XBOX 360 controllers to show.

dmcb, 2017-11-10:
@danvtim what exactly did you do to fix this? It's unclear what you mean by "enable permission for the extension to run the the security prefs". Thanks!

nour-s, 2017-11-10:
@dmcb Check this out
https://github.com/360Controller/360Controller/issues/495#issuecomment-343284644

I fixed my problem using that.

Melechim, 2017-11-19:
I have a 2017 MacBook Pro, and I have two different controllers I am trying to get to work. My xbox 1 controller will rumble and will light up (if I turn it on). It will not show my controller on there at all. what do I do?

iampaulmata, 2017-11-21:
It took me  a little while to figure out what @danvtim meant about the permissions. When you install the latest package, have your System Preferences open to the General tab under Security & Privacy. You will see a notification that says "System software from developer "Drew Mills" was blocked from loading." and there will be an "Allow" button. Click on that (you may have to unlock and enter your admin password first) and reboot or unload and reload the kext and you should be up and running. It worked for me!

[image]

FranticRain, 2018-04-02:
PowerA Xbox One controllers are incompatible with macOS 10.11+ thanks to a rewrite of the macOS USB kernel. We cannot resolve this issue. It is a bug in Apple's core OS code.

closed 2018-04-02 by FranticRain