360Controller/360Controller #548: Xbox one controller won't work with Borderlands 2

cjrexach, 2017-11-05:
Using your driver with a wired Xbox One controller has worked perfectly with a couple of emulators but when I try to play Borderlands 2, it won't recognize the controller.  I even tried the "Spoof Xbox360" option and it still wouldn't recognize it. Appreciate any help I can get.

PokeCaptain, 2017-11-06:
Borderlands 2 works absolutely fine for me with [this](http://pdp.com/en/shop/legendary-collection-sound-of-justice-wired-controller-for-xboxone-pc) third-party controller. It does freeze up and crash if I start the game before I plug in the controller through. If I plug in the controller before starting it up, it works fine. Note that I'm using the Steam version, and not the one from the App Store.

FranticRain, 2018-04-02:
The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

closed 2018-04-02 by FranticRain