360Controller/360Controller #551: xbox one controller not detected on the driver

emperorm2, 2017-11-10:
The xbox controller driver does not recognize the controller

My set up is
・MacBook Air 2017 with osSierra Version 10.12.6
・Wired controller for Xbox one by pdp
・xbox controller driver -> 360Controller 0.16.5
My issue is
・driver does not detect any device
・Mac does recognize the controller in the system profiler
・light does not come on even if plugged in
・firmware updated to the newest version
My question
・What is the problem(s)?

Melechim, 2017-11-19:
I'm have a MacBook Pro 2017. I can get my controller to light up but I can't get the driver to recognize
either controller I try to plug in.

johnflan, 2017-11-25:
Similar issue here running high sierra 

[image]

Melechim, 2017-11-26:
I got it to work but it only worked until I turned my comp off. I tried to go into my security preferences and turn it back to allow but its not working anymore

CryingMagma, 2017-11-26:
Have you tried using a different usb cable to plug it in, sometimes the USB cable won't work and the solution is to just use a different one.

Melechim, 2017-11-26:
It worked for a couple of days I was using the data cable I got with my charger pack for the xbox 1 I don’t understans why it just quit working

reteps, 2017-11-30:
go to advanced and check "enable driver". This fixed it for me.

emperorm2, 2017-12-01:
Enabling the driver in the advanced section did not work for me.

Melechim, 2017-12-01:
Yeah I’ve done all of that I don’t know what happened

FranticRain, 2018-04-02:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

closed 2018-05-17 by FranticRain