360Controller/360Controller #553: Original XBOX 360 USB controller not detected.

dipunm, 2017-11-11:
I have tried the following steps:
- unplug and replug the controller
- enable driver from Advanced tab (not sure why it was disabled)
- disable and re-enable driver.
- uninstall drivers and install 0.16.4 (because my controller is old and 0.16.5 says it supports only new devices)

Here is my driver report:

[image]

and my OS:

[image]

DoctorAngelo, 2017-11-18:
I have macOS sierra high and xbox 360 controller but it doesn't work. Please help.

sterpe, 2017-11-19:
FWIW, I have been able to get an [X-Arcade Stick w/PS2->Xbox360](https://shop.xgaming.com/collections/adapters/products/x-arcade-xbox-360-adapter) adapter working on High Sierra (10.13.1) via USB 2.0.  Installed 0.16.5 `.dmg`, approved drivers signed by (Drew???) in Security Preferences, reboot, enable drivers in System Preferences (have to do this everytime, for some reason).

@dipunm My thought would be first do upgrade to 10.13.1.  Also maybe try to connect to older USB 2.0 hub and then into machine, if you have one lying around, rather plugging controller directly into builtin USB3.

[image]
[image]
[image]

dipunm, 2017-11-22:
I got it working thanks to @sterpe. I updated my OS, and the application. The one thing that I think was key was the privacy settings step. For others: open System Preferences and go to "Security & Privacy"; then go to the general tab.

[image]

Theres a button to allow installation of drivers from Drew. Allow it, and restart. Voilla!
Even works without a USB hub for me. :) Thanks.

closed 2017-11-22 by dipunm

solomani, 2019-05-04:
Thank you, this fixed my issue as well.