360Controller/360Controller #554: Is this just a rebuilding from the original version?

lethelimited, 2017-11-12:


Cruplezone, 2017-11-19:
no

Deph0, 2017-11-19:
From the [Readme](https://github.com/360Controller/360Controller/blob/master/Readme.md#im-using-a-driver-from-the-tattiebogle-website)
>###  I'm using a driver from the Tattiebogle website
> 
> The Tattiebogle driver is NOT the same driver as this Github project. We do NOT support that driver. Under NO circumstances will we support that driver. If you download the latest version of this driver from the releases page we will do our best to help you out. This driver will install over the Tattiebogle driver. You don't have to worry about uninstalling the Tattiebogle driver first.

FranticRain, 2018-04-02:
Don't make issues without even reading the README. Come on, dude.

closed 2018-04-02 by FranticRain