360Controller/360Controller #561: 360controller doesnt recognize my XBox One controller

Portal2020, 2017-11-27:
So i bought myself a wireless XBox One controller today. I downloaded the newest version of 360controller and everything is working fine, but it doesnt recognize my controller. Can anyone help me with that?

reteps, 2017-11-30:
go to advanced and check "enable driver". This fixed it for me.

Portal2020, 2017-12-01:
did that, but it still doesnt recognize the controller

Nastrella, 2017-12-02:
Hi, I'm following as I have the same issue. I'm on OS 10.13, I ticked “enable driver“, turned the computer on and off, but it still “No device found“.

ppahv, 2017-12-02:
Hi,
I've got the same problem via Bluetooth with my Xbox One S Controller. Whereas 360controller recognises it when plugged with a USB wire, 360controller doesn't detect it when it's paired via Bluetooth.
Nonetheless, the controller works like a charm via Bluetooth, and I can use it and map it however I want in games and with the Enjoyable app. The only problem is that I can't map the two triggers RT and LT, since they're not detected when connected via Bluetooth.

KorvinSzanto, 2017-12-07:
Try the steps listed here: https://github.com/360Controller/360Controller/issues/566#issuecomment-349858763

Nastrella, 2017-12-26:
It's finally working for me with the steps from #566, the dialogue appeared :)

closed 2018-03-29 by FranticRain