360Controller/360Controller #563: Original Xbox One controller not working

glebnerzhin, 2017-12-04:
I'm using Mac OS X El Capitan with 10.11.6
Running the 0.16.15 driver
Product ID:	0x02d
Vendor ID:	0x045e  (Microsoft Corporation)

So, it won't recognise at all my controller, I tried enabling and disabling the driver and also uninstalling it and installing it again. What else can I do, is there any known issue or workaround? Thanks!

MarioGiancini, 2017-12-11:
In El Capitan and Sierra  (10.11 & 10.12) I was able to my Xbox one controller working with a USB cable. 

But in **High Sierra** it doesn't seem to find the controller, even with ensuring `Enable Driver` is checked in the `Advanced` tab.

[image]

sterpe, 2017-12-11:
Did you authorize the drivers in System Security Preferences after install?

glebnerzhin, 2017-12-11:
@sterpe How would I do that?

FranticRain, 2018-04-02:
System Preferences -> Security and Privacy -> General

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain