360Controller/360Controller #564: play gaming accessories xbox 360 game pad "no devices found"

srosin, 2017-12-04:
play gaming accessories xbox 360 game pad "no devices found"

Wired Controller for Xbox 360:

  Product ID:	0xfafa
  Vendor ID:	0x24c6
  Version:	1.00
  Serial Number:	06EA09D4
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14423000 / 14
  Current Available (mA):	500
  Extra Operating Current (mA):	0

KorvinSzanto, 2017-12-07:
Try the steps listed in this comment: https://github.com/360Controller/360Controller/issues/566#issuecomment-349858763

srosin, 2017-12-07:
there was no dialog in "System Preferences/Security & Privacy/General" @KorvinSzanto

Nastrella, 2017-12-07:
Same as serosin, no dialogue showed up.

[image]

xboxjoey, 2017-12-09:
I seem to be having issues with the same model of controller. Curious if it is High Sierra or the model of controller?

https://github.com/360Controller/360Controller/issues/567#issuecomment-350401738

closed 2018-04-03 by FranticRain