360Controller/360Controller #566: Xbox One controller not recognized on iMac, but recognized on macbook pro

KorvinSzanto, 2017-12-07:
I'm trying to upgrade from a macbook pro to an iMac and the controller setup I was using won't work with it. Here's the connected controller according to my iMac's system information:

| | |
| ------- | --------- |
 | Product ID:	| 0x02ea  |
 | Vendor ID:	|0x045e  (Microsoft Corporation)
 | Version:	|3.01
|  Serial Number:	|30324F5830313530353834363335
|  Speed:	|Up to 12 Mb/sec
|  Manufacturer:	|Microsoft
 | Location ID:	|0x40110000 / 5
|  Current Available (mA):	|500
|  Extra Operating Current (mA):	|0

I have a usb cable plugged from the controller straight into my iMac and I have the "Enable Driver" checkbox checked. When I plug in the controller, it vibrates but it doesn't light up. If I look at my system report I can see the controller is plugged in and recognized. When I open the Xbox 360 Controllers pane in my system preferences, it says "No devices found" no matter what I do.

If I unplug the USB end of my cable from my iMac and plug it into my macbook pro running the same version of macos and 360controller, the controller shows up right away and gets a steady light.

Does anyone have any idea what I might be doing wrong?

sterpe, 2017-12-07:
I don't work on this project merely follow it and this may be an obvious and dumb question but did you reinstall the driver on the iMac or was the driver installation copied to the new machine via Migration Assistant/Time Machine/etc?

KorvinSzanto, 2017-12-07:
@sterpe It was originally copied via a time machine restore. I uninstalled it and reinstalled it fresh :S

Here's what I get in console.app when I connect the controller:
```
default	19:45:36.532809 -0800	icdd	#ICDebug - 23:{ICWiredBrowser.m} (USB Device first match)
default	19:45:36.533276 -0800	icdd	#ICDebug - 388:{ICWiredBrowser.m} (11 USB Descriptions Managed)
default	19:45:36.534092 -0800	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0x0/0x0/0x0 - 0x41130000 - ICDeviceDescriptionSUQuery)
default	19:45:36.574965 -0800	icdd	#ICDebug - 215:{ICResourceManager.m} (33303332-3446-3538-3330-333133353330|Controller|MANUFACTURER:Microsoft;MODEL:Controller|SW=FALSE|)
default	19:45:36.575940 -0800	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0x0/0x0/0x0 - 0x41130000 - ICDeviceDescriptionAdded)
default	19:45:39.934855 -0800	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	19:45:41.318924 -0800	icdd	#ICDebug - 51:{ICWiredBrowser.m} (USB terminate)
default	19:45:41.318973 -0800	icdd	#ICDebug - 344:{ICWiredBrowser.m} (--> USB Terminate)
default	19:45:41.319112 -0800	icdd	#ICDebug - 363:{ICWiredBrowser.m} (Reclaiming Resource: Controller)
default	19:45:41.319148 -0800	icdd	#ICDebug - 504:{ICDDMessageCenter.m} (-Rem Controller - 0x0/0x0/0x0 - 0x41130000 - ICDeviceDescriptionUndefined)
default	19:45:41.319149 -0800	icdd	#ICDebug - 371:{ICWiredBrowser.m} (10 USB Descriptions Managed)
default	19:45:41.319170 -0800	icdd	#ICDebug - 373:{ICWiredBrowser.m} (<-- USB Terminate)
default	19:45:41.539788 -0800	icdd	#ICDebug - 23:{ICWiredBrowser.m} (USB Device first match)
default	19:45:41.540423 -0800	icdd	#ICDebug - 388:{ICWiredBrowser.m} (11 USB Descriptions Managed)
default	19:45:41.541207 -0800	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0x0/0x0/0x0 - 0x41130000 - ICDeviceDescriptionSUQuery)
default	19:45:41.581447 -0800	icdd	#ICDebug - 215:{ICResourceManager.m} (33303332-3446-3538-3330-333133353330|Controller|MANUFACTURER:Microsoft;MODEL:Controller|SW=FALSE|)
default	19:45:41.582339 -0800	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0x0/0x0/0x0 - 0x41130000 - ICDeviceDescriptionAdded)
default	19:45:41.589426 -0800	kernel	CoreStorageGroup::handleOpen(<private>) called, client <private>, access 3, state 5, client is a lv = n.

```

Kinda odd that the usb connection terminates after connecting, no?

KorvinSzanto, 2017-12-07:
I found the solution! If you are having the same issue I was having, try these steps:

1. Open "System Preferences"
2. Go to "Security & Privacy"
3. Click the "General" tab
4. Notice at the bottom of this dialog it says "Some system software was blocked from loading". Click the "Allow..." button and allow from author "Drew Mills", that's @FranticRain 

That's all! Now the controller will show up!

sterpe, 2017-12-07:
Cool, drivers and kernel extensions and such always have to be reinstalled/ recompiled for a new machine.  +1

closed 2017-12-07 by KorvinSzanto