360Controller/360Controller #567: Doesn't work on high Sierra ?

Akrosid, 2017-12-08:
Hello i have a Xbox 360 Wireless controller, it work really good on my pc.
I installe this driver but my controller is not reconize ? It is because i'm on High Sierra ?

xboxjoey, 2017-12-08:
I am also having issues getting the controller to be recognized on High Sierra (10.13.1)

The controller shows under 'About this Mac > System Report > USB':
The brand is "PLAY"
```
Wired Controller for Xbox 360:

  Product ID:	0xfafa
  Vendor ID:	0x24c6
  Version:	1.00
  Serial Number:	06EA2076
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14200000 / 8
  Current Available (mA):	500
  Extra Operating Current (mA):	0
```
- I have installed Version 0.16.5 of the Xbox 360 Controller Driver.
- I went to Security and Privacy > General and checked the box to "allow Drew Mills app"
- Under 'Advanced' options of the Preference Pane, I've checked "Enable Driver"
- It still displays "No devices found" in the Preference Pane
- I've restarted my computer

- I've tried games with both the driver enabled and disabled. And with games that that list the Xbox controller as a selectable option.

- When starting up the computer, the Xbox button blinks a few times, and then stays off
- Pressing the button after that does nothing

Any ideas? Is this particular controller not supported? Of is it a High Sierra issue? I was told when purchasing it, that any Xbox 360 controller would work. 

Thanks to anybody who can help :)

vreyens, 2017-12-11:
It works on High Sierra, just delete the program, restart your mac, then download, restart, and try again, it worked for me and all I have to do now is figure out why I cant turn left or right using right analog stick

xboxjoey, 2017-12-11:
Hi vreyens,

Thanks for the suggestion. I have tried everything including uninstalling and reinstalling with restarts.

Are you using the same model controller?

vreyens, 2017-12-15:
No I am using a wireless Xbox One controller(comes with the console) and a micro usb cord that came with something else that I have, but I had the same problem you had. But if you can I would recommend using a wireless controller with batteries and then just plugging it in, because the computer will shut off USB ports if anything is using too much power.(Which I found out was why my controller didn't work at first) When this happens you have to unplug the controller and use one with a charged battery, or at least that's what I have to do.

phenemann, 2017-12-30:
I'm using a wireless Xbox 360 Controller with the usb dongle and having the same issue... What's really weird is that my driver disable itself (should it be like this?).
I have also tried multiple versions. No lucky so far.

if there's anything I could help or information I could collect to solve this issue, please tell me.

tcash22, 2018-02-01:
@phenemann did you figure out this Xbox360 wired controller issue? I'm having the same problem

phenemann, 2018-02-01:
I actually don't have a wired controller to test it.

closed 2018-04-03 by FranticRain

vreyens, 2018-04-12:
Did everyone get this figured out for themselves? I figured it out I just
forgot to respond to you.

vreyens, 2018-04-13:
> @vreyens <https://github.com/vreyens> I didn't had the time to test this
> release. For sure I will in the next couple hours. Thanks man

No problem, if you find anything don't hold out on responding I might end
up needing it sometime

phenemann, 2018-04-13:
@vreyens I'm sorry it took me this long. The 360 wireless controller, correct? if so, I can't really test it, since my controller is a wireless one. I believe @tcash22 has a wired controller to confirm this fix.
Thanks man.

vreyens, 2018-12-02:
Hey, can you sign this petition? http://chng.it/hyPKmhkSLD

Thanks! - Stephen