360Controller/360Controller #569: how to uninstall on mac

Enchman69, 2017-12-12:
hi im wondering if you have any directions on how to uninstall this program from my mac? its just not working correctly

xboxjoey, 2017-12-13:
Go to System Preferences and click on the 'Xbox 360 Controllers' Preference Pane and then click 'Advanced' and you will see the button to uninstall. It's not working for me either, but I'm thinking it might be the model of my controller...

Enchman69, 2017-12-13:
i did so, however when installing it left a drive called "VM" thats completely messing up my system
...
[image]

Enchman69, 2017-12-13:
i did so but its not removing this drive called "VM" which is messing with my mac pretty badly 
...
[image]

xboxjoey, 2017-12-13:
Weird. I don't have that. I'm on macOS 10.13.1 with the software still installed. 

Does VM stand for virtual machine? Are you running another OS on your computer? I'm not very knowledgable, just making guesses...

Adansix, 2017-12-27:
@Enchman69 
VM stands for Virtual Machine. I'm guessing you emulate Windows on your Mac or whatever...

FranticRain, 2018-03-29:
https://github.com/360Controller/360Controller#uninstallation

The VM drive isn't created by the driver.

closed 2018-03-29 by FranticRain