360Controller/360Controller #571: Xbox One controller doesn't work half the time

XrayXout9, 2017-12-14:
My recently given Xbox One Face-off Controller is a wired only controller. Product is made by pdp

when i first plugged in the controller, it detected the controller as a Xbox One controller but, the controller layout didn't show that it was working so, I go and disable the driver then re-enable the driver and that seems to work but, the Xbox Logo still won't light up at all and when I press the button the the back of the controller, it disconnects COMPLETELY and, I have to disable and re-enable the driver again in order for it to work. 

All i want is too see the xbox logo light up so I KNOW that it is plugged in and working. 

I am currently on(and can never upgrade from) OS X El Capitan ver. 10.11.6

H10Maxwell, 2017-12-22:
Did they ever get back to you to fix this?

sterpe, 2017-12-22:
I don't know, but it seems like your controller is working most? all the time and the issue is that the logo light doesn't come on?

If that's the case you should change your bug report to something like 'PDP-built Xbox One controller logo light doesn't light up even when controller is connected and working.'

And then file a separate issue for the situation around pressing the button on the back of the controller (which button?) and it disconnects completely and needing to re-enable the driver again.

The more specific you can make your issues in this repo (or any open source github repo), the better the chance of someone being able to look at it and do something, but unfortunately no one is going to be able to go out and buy this specific controller to investigate. This repo sees a ton of 'my controller doesn't work' type of reports so, it helps to be very specific about what is working and what's not working, and, this is critical, with well explained steps to reproduce.

In any report you should also provide the USB device report for the device from your About this mac/system report.

FranticRain, 2018-04-02:
PDP Xbox One controllers are incompatible with macOS 10.11+ thanks to a rewrite of the macOS USB kernel. We cannot resolve this issue. It is a bug in Apple's core OS code.

closed 2018-04-02 by FranticRain