360Controller/360Controller #572: Sistem crash after unplug

NachoFornier, 2017-12-14:
Hello, I'm having a problem, when I unplug the xbox 360 wireless controller, if I close the Mac or shut down the mac system crash and reboots. If I do not turn off or close the mac, I mean, i still use it, no problem occurs.
I have the latest version of the driver installed and I am running the latest version of mac os high sierra on a macbook air 2017.
Sorry for my bad English, I am using a translator. Greetings Ignacio

BillKills974, 2017-12-28:
Hello, I got the exact same problem on macOS High Sierra. Kernel panics on shutdown/sleep after having used and unplugged an xbox 360 wireless controller. System runs fine if the device has not been plugged since boot.

Glutnix, 2018-01-01:
I also am using a MacBook Pro (Retina, 15-inch, Mid 2014) running High Sierra 10.13.2 with 360Controller 0.16.5, and I experience a kernel panic soon after not using the Xbox 360 Wireless Receiver. 

I don't have to unplug it for the panic to occur, I just have to stop using the controller temporarily, for about 5-10 minutes.

Also, closing the laptop will trigger a panic on reopening it, no matter if the receiver is attached or not.

NachoFornier, 2018-01-22:
you're welcome :)

FranticRain, 2018-04-02:
This crash is caused by a messy dismount that occurs whenever the computer goes to sleep or shuts down while the controller is still wirelessly connected to the receiver. It is a bug with macOS Sierra and later. There is no known fix. Manually disconnecting the controller prevents the issue.