360Controller/360Controller #573: Enabled driver and program crashes

TBatsS, 2017-12-16:
hi

I have installed the latest driver and also some recent ones to try and help me but I'm having no luck at all.

basically I installed 360controller 0.16.5 and restarted my MacBook pro. i have allowed access in security settings also. then i go to preferences and click 'the 360 controller' icon at the bottom. and as soon as i connect the controller and its picked up buy the software, the application just crashes/closes and i get the report crash option. this happens every time and i have also tried playing a game but it doesn't work. i have tried other drivers and also some other work arounds but i have no luck.

please help :(

FranticRain, 2018-04-02:
What controller are you using and what are the product and vendor IDs of the controller? https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain