360Controller/360Controller #574: 3rd Party Wired Xbox One Controller Not Recognized

H10Maxwell, 2017-12-22:
Product ID:	0x0160
  Vendor ID:	0x0e6f

I have PDP wired Xbox One controller that shows in the About This Mac > System Report screen, but isn't found in the System Preferences > Xbox 360 Controller screen. I'm very inexperienced with coding so I hesitated to go further without help. I am running OSX Capitan 10.11.6

sterpe, 2017-12-22:
See https://github.com/360Controller/360Controller/issues/571#issuecomment-353518154 for how to improve this report for greater visibility and higher probability that someone can eventually address it.

sterpe, 2017-12-22:
Possibly see https://github.com/360Controller/360Controller#adding-third-party-controllers

FranticRain, 2018-04-03:
PDP and PowerA Xbox One controllers are incompatible with macOS 10.11+ thanks to a rewrite of the macOS USB kernel. We cannot resolve this issue. It is a bug in Apple's core OS code.

This device will still be added, as it should still work with El Capitan.

closed 2018-04-03 by FranticRain