360Controller/360Controller #575: xBox One controller conflicts with MFi gamepad

JulyIGHOR, 2017-12-25:
I have tried to use Nimbus game controller + xBox One game controller (via this 360Controller driver)
And there is issue, both gamepads starting to control first player in the game.
Please add option to define 360Controller driver index of controller to 2, 3 or 4

FranticRain, 2018-03-29:
That's not something that the driver controls. That's a bug in the game you are playing. No fix possible.

closed 2018-03-29 by FranticRain

JulyIGHOR, 2018-03-29:
It not the game, I tried many, it happens for all games.

FranticRain, 2018-03-29:
Does it work in the preference pane correctly? It it's correct in the preference pane, then it's the games.

FranticRain, 2018-03-29:
While the controllers likely gave you the impression that they are assigned a role because of the indicator lights, that is simply a cosmetic effect. In the case of the Xbox 360, the operating system assigns each controller a number. Unfortunately, in macOS, this isn't done by the operating system, and is instead the responsibility of the game. The problem that you're running in to is that the means a game uses to access mFI controller is different from the one used to access non-mFI controllers. The games don't properly handle this and unfortunately it means that you have that issue. The driver has no control over who is who in this controller lineup, so unfortunately, there is nothing that we can do.

JulyIGHOR, 2018-03-29:
Yes, I understand. Thanks for explanations.