360Controller/360Controller #577: Wired Xbox 360 controller: right stick button stopped working

DrPuppet, 2017-12-30:
Hi all: The subject basically says it all. Everything is working fine except the right analog stick button. I noticed it in Divinity: Original Sin Enhanced, but when I opened the System Preferences panel, it wasn’t registering there either. It works in Boot Camp just fine, so I don’t think I’ve busted it on the controller. Just updated to latest release, but issue persists. Running OS 10.12.6. Any suggestions? Thank you!

DrPuppet, 2017-12-30:
Update: Disabled and re-enabled driver, which got the right-stick button working again, but noticeably less consistently than it used to. The input recognition seems “sticky.” I’ll try again in Boot Camp tonight or tomorrow to double-check that it is still working as well as it used to there.

FranticRain, 2018-05-23:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-23 by FranticRain