360Controller/360Controller #580: Input Lag Of Buttons

EmmettShaffer, 2017-12-31:
I have been trying to play Roblox with my wired Xbox One controller on my Mac Book Pro, (Don't judge me please) and I would sometimes press a button down but it would sometimes act like it wasn't pressed and not do anything in game or it would act like it has pressed longer then it actually has, if anyone knows how to fix this please reply.

Just incase the specs for my Mac Book Pro are... 
[image]

FranticRain, 2018-04-02:
It might be hard to test with the infrequency of the issue, but is this issue also displayed in the System Preferences controller test? My guess is that it isn't, which would make this a game specific issue, and not something that we could fix.

FranticRain, 2018-05-23:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-23 by FranticRain