360Controller/360Controller #581: Unrecognized controller with USB hub on new macbook pro

fguerin, 2018-01-02:
Hi guys,

Few months ago I bought the new mac bookpro (with only 2 USB-C ports) and i'm trying to connect my Xbox 360 controller. As there are only USB-C ports, I bought the apple official hub (with 1 USB, 1 HDMI and 1 USB-C ports). I installed the 0.16.5 driver version and when I connect the controller, there are green lights for less then 1 second and nothing. 
I checked in my Hardware and there are only HUBs.

Doesn't someone had the same issue? Thanks.
Florian

[image]

lordzsolt, 2018-01-09:
Trying to do the same thing as well, can't seem to make it work :(

FranticRain, 2018-04-02:
Sorry. We only work within the parameters of what the operating system provides to us. If the operating system doesn't show us the USB devices, we can't do anything about it. I can recommend you try a different cable to see if the cable you are using is a charge cable and not a data cable, but it is impossible for us to provide a fix for this issue.

closed 2018-04-02 by FranticRain