360Controller/360Controller #582: controller not connecting

wiyao, 2018-01-03:
I did every thing correct, I installed it and every thing but when I connect it to the computer and when I went to it in settings the controller didn't connect. please help

FranticRain, 2018-04-02:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#wired-xbox-360-controllers

courtsimas, 2018-04-03:
Same. my 360 controller shows up as a 'vendor specific device' in the system report -> usb section, but the driver's page in system preferences still doesn't recognize it at all.

FranticRain, 2018-04-03:
Make a new issue, and please provide your PID and VID.

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain