360Controller/360Controller #584: Can't Connect Xbox One Controller To Mac

queentatertot, 2018-01-06:
I have the xbox 360 controllers drive installed and I'm using a microsoft xbox cord to link it to my computer, so it isn't a charge cable. The controller vibrates when connected, but isn't recognized even with the controller pane open. 

Here's it's PID and VID, just in case: 0x02d1
  Vendor ID:	0x045e  (Microsoft Corporation)

FranticRain, 2018-04-02:
Have you allowed the application's signing certificate in the Security section of the preference pane? This controller should be supported.

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain