360Controller/360Controller #587: Which controllers definitely work on high Sierra?

11744, 2018-01-09:
As it seems its not possible to use 3rd party controllers on high sierra, which offical microsoft controllers definitely work? Wired or wireless?

Thank you

RobertMcK, 2018-01-13:
I've just successfully got an official xbox 360 wired controller on high sierra to work.
Had to allow the software through my firewall and then check the box that says "pretend to be an xbox controller" before it'd work though.

Cheers for the software guys! Much appreciated

thirdrockfromthesunshine, 2018-01-25:
really struggling to get a new xbox one s controller to work on high sierra on new macbook pro 2016 with touch bar. iv tried everything. via bluetooth i cam get it to trigger some butons but they are all out of order even when used joystick mapper... iv tried the wired method using github drivers and it doesnt even recognise a pad as being connected. I really want to play Fortnite using a pad. iv spent 80 pounds on control pads in the last month 1st buying a steelseries nimbus which doesnt bind correctly and now this new xbox controller that is only visable through bluetooth but doesnt work in game even though the game has controller settings for xbox controllers... someone please help

FranticRain, 2018-04-02:
There are a plethora of issues within this single issue. First, read the README, it will help a lot with this. 

Second, there are some game specific issues here. The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

closed 2018-04-02 by FranticRain