360Controller/360Controller #589: Mac OS driver support for 3rd party Xbox 360 wireless receivers

pkc123, 2018-01-17:
Can anybody confirm this MAC OS driver will work with 3rd party Xbox 360 wireless receivers similar to following?
https://m.ebay.com/itm/NEW-PC-Wireless-Controller-Gaming-Receiver-Adapter-For-Microsoft-XBOX-360-White/382332421930?epid=1101622271&hash=item5904c9c72a:g:f9UAAOSwfjRaDkMS

sevdestruct, 2018-01-17:
I've used this with a 3rd party receiver. Funny thing was that this Mac unofficial driver recognized it better than windows which requires manually selecting the Microsoft driver.

pkc123, 2018-01-17:
Thanks a lot. I’m also thinking of giving it a try. BTW how about consistency of connectivity of your receiver. Have heard most of the time people complain about connecting and disconnecting of such receivers often.

sevdestruct, 2018-01-24:
Never had an issue.. but honestly I only ever tested one 3rd party receiver… I can't speak for them all.

pkc123, 2018-01-24:
Thanks. I'm also going to try one of those. Will update the result then.