360Controller/360Controller #590: Switching triggers and sticks?

cahillja20, 2018-01-23:
Thank you for providing this availability to use the xbox controller on the mac! I have been trying to play Fortnite, and I was able to remap almost everything to get it to completely work. However, turning can only be used by the left and right triggers, and scoping is pushing down on the left stick and shooting is pushing down on the right stick. Any way this can be fixed? I looked for a way to be able to change it in the system preferences but it doesn't seem like there's an option.

Thanks.

sydnidaren, 2018-03-23:
I'm having the same issue...I tested all my buttons on system preferences and they all seem to be working, but when I try to play Fortnite the B button works as the A button and I can't use the right joystick...HELP

sydnidaren, 2018-03-23:
I was able to get it to "pretend to be a 360" remote in system preferences and it's now working!!

FranticRain, 2018-03-29:
Swapping triggers and sticks simply doesn't work. Use the previously mentioned checkbox, as it is the only solution we can provide.