360Controller/360Controller #592: 360Controller

Sethmr, 2018-01-28:
I have Mac High Sierra 10.13.2. I just purchased an Xbox Elite Controller with the assumption that I could connect it to my mac. Why did I trust some random guy I didn't know at best buy *facepalm*. I had previously tried to get my Xbox One Controller working with it to no avail. Neither controller has ever stayed connected to the computer. I can work my way around the terminal if I need to do anything, but there just seems to be little to no information on what I should do. I figured you guys could point me in a direction if you were aware of any.

Hansaplast, 2018-03-11:
Did you try connection the Elite controller with a USB cable to your Mac? Seems to work just fine on my Mac (10.13.2 as well).

Sethmr, 2018-03-11:
@Hansaplast I did try this. Is there a specific cable I need maybe? I was using the original cable that came with my xbox, though it is very old, so I wouldn't be surprised if it didn't work properly. I tried several other cables lying around, but I don't know for sure they were data cables.

Hansaplast, 2018-03-12:
You will need a data cable. I used the one that came with my controller and I tried a random cable that I had laying around - both worked. I know though that there are some really cheap cables for charging only, I just do not seem to own one hahah :-)

Hansaplast, 2018-03-12:
Note that the cable was part of my first XBox One (late 2013) so that cable wasn't exactly new either ;-) 

Just occurred to me that more recent versions of macOS need their applications and drivers to be signed. Usually Gatekeeper allows you to work around this when the application is started for the first time. Not sure if it's relevant for this tool, but you may want to uninstall it and/or start it and check in your System Preferences and see if it's show under the Security and Privacy option (tab: General) and see if it shows the application there so you can add it to the exception list (thus allowing the application to run).

FranticRain, 2018-04-02:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.