360Controller/360Controller #595: Xbox One S Bluetooth Controller not recognized on Sierra

tianyi5309, 2018-02-02:
My bluetooth controller is not recognized by 360Controller on Sierra.
Here's the rundown:
1. Controller connects via bluetooth normally, I can play all my games.
2. However, there's no rumble support. That's why I'm using 360Controller.
3. If controller is connected via bluetooth, 360Controller does not detect it.
4. If controller is connected via USB (bluetooth off), then 360Controller will detect it, and everything works fine.
5. I have SIP disabled.

Do you have any idea how to fix this?

MaddTheSane, 2018-02-02:
The plug-in needed for Bluetooth support isn't included in the most recent release yet. As the Bluetooth controller is handled by the OS itself, only the force feedback driver is needed. If you want, you can build the force feedback driver yourself.

tianyi5309, 2018-02-02:
Thanks for the clarification!
How do I build the force feedback driver?

tianyi5309, 2018-02-02:
So I've been playing around with Xcode a bit, but I still have no idea how to build the driver and install it. Does the driver come with the full install? Otherwise, how exactly do I build it/ can you release a built driver? Thanks!

MaddTheSane, 2018-02-02:
1. Build `XboxOneBluetooth`
2. Locate where Xcode built the kext.
3. Set the proper permissions for kexts: `sudo chown -R root:wheel XboxOneBluetooth.kext`
4. Move the kext to `/System/Library/Extensions`. 

Note that this will only work if you have SIP disabled. A new release would be needed if you wanted to keep SIP enabled.

tianyi5309, 2018-02-02:
Do I need Xcode 6.4 to build it?

MaddTheSane, 2018-02-02:
Any version of Xcode can build the Force Feedback drivers. The Bluetooth kext is only there to point the OS to the Force Feedback driver.

tianyi5309, 2018-02-02:
Thanks, it works now! However, the rumble seems to be rougher and less granular than the ones on Windows.

closed 2018-02-02 by tianyi5309

melvyn2, 2019-01-15:
I've compiled this and installed `XBOBTFF.plugin` and `XboxOneBluetooth.kext`, connected it (regular support works), but nothing shows up in the driver panel. The drivers ARE loaded.

FranticRain, 2019-01-15:
1. Bluetooth controller stuff isn't supported. Not exactly sure what you expected by posting here.
2. It never showed up in the preference pane. That's by design. Bluetooth controllers aren't supported by the driver. All that stuff was to add Force Feedback to those controllers, not attach them to 360Controller in any way.

melvyn2, 2019-01-15:
Thanks for explaining, probably should have search more before posting.