360Controller/360Controller #598: macOS High Sierra

magresh1, 2018-02-08:
The mac recognizes the controller under About this mac> System report> USB but the software does not recognize it and says there is no device available. Please for the love of god come out with an update for bug fixes on High Sierra so I can play Fortnite lol

shangles, 2018-02-08:
You did not specify what controller you're using, or whether it's wired or wireless.  I'm running High Sierra and my Xbox 360 wired controller works fine.  Try disabling/enabling the driver once the controller is connected and see if that helps.

magresh1, 2018-02-08:
Ya I tried that it’s not working with my Xbox one controller. I’ll try it
with the Xbox 360 controller thanks

lipasite, 2018-02-08:
XBOX360 Wireless Controller dont working. A try to configure it under XBOX360Controller System Menu Driver, but controller dont recognize, at controller list - empty.

talusproteus, 2018-02-08:
I am also unable to get my controllers to work. Both of them are wired Xbox 360 controllers, but one is an original Xbox controller and the second is made by Afterglow (for Xbox 360). As the OP stated, my mac recognizes the controller under About this mac> System report> USB, but the software does not recognize either controller and says there is no device available. 
[image]
[image]
[image]

lglucas, 2018-02-12:
I'm having the same issue here.
I have two xbox 360 controllers, one is wired, and both can't work anymore.

lipasite, 2018-02-13:
it looks like the developer left us..

lglucas, 2018-02-13:
Actually, I made it.
It was easy to solve, FireVault was blocking it. 

Just install normally and then go to SystemPreferences>Security & Privacy> Press "Allow" on the bottom right. (Check the developer name)

lipasite, 2018-02-13:
Did you mean "FileVault"? (not "Fire")
FileVault is disabled on my system by default. But still not working and not recognize.

kurochief, 2018-02-14:
Hello guys! I think I solved it.
Try to go to Xbox 360 controller settings and in the Tab "Advanced", click on Uninstall. Then restart your mac. After that, install the version 0.16.65, restart again. Check on settings > privacy and security > general > allows Drew Mills app.

After this, connect your joypad. It should work.

fortnite123, 2018-02-16:
Having the same issues. Shows in system report that there is a controller connected, but doesn't show up on the driver, tried with both xbox one and a wired 360 controller. ]

Need a bug fix for fortnite.

lolek, 2018-02-17:
The same problem here. Clicked to allow. Did not helped.
10.13.3 (17D47)
Xbox One Wired

lolek, 2018-02-17:
Installer older version (Version 0.16.2) and it is working!!! 👍

Gordonphilips, 2018-02-23:
I’m having the same problem with a wired xbox360 controller. It shows up in usb info, but I don’t get the controller icon in preferences. I’ve downloaded the driver but it won’t install. The Drew’s app  does not show up in my General preferences, So I can’t  manually authorize it. Please help

pauliharman, 2018-02-28:
Same here. Yes I have allowed the Drew Mills stuff and rebooted. Wired 360 and wireless One controller connected via MicroUSB both show up in system preferences but neither are found in the Xbox 360 preferences dialog.

FranticRain, 2018-04-02:
Since this issue has morphed and changed and deviated from any single issue, I am going to close it. If you still have an unresolved issue, open a new issue.

closed 2018-04-02 by FranticRain

MiaSparx, 2018-05-22:
Dear @FranticRain 

I appreciate your help. I'm not so sure this issue morphed and changed. It appears instead users like myself tried troubleshooting various options and suggestions in your absence. Did we ever land on a fix for using Xbox One original controllers with MacOS High Sierra? As it stands I can't enjoy playing Fortnite Duos with bae. I mean I could, but it would suck and I would hate it. Can we reopen/revisit this topic?

FranticRain, 2018-05-22:
@MiaSparx If you could make a new issue so that I could help fix your individual issue, that'd be excellent. The Xbox One controller via 360Controller works just fine with macOS High Sierra. I'm doing it right now.