360Controller/360Controller #600: x

jackman123, 2018-02-09:
mine does not recognize my controller nor is my blue tooth connecting to my controller i have a xbox one controller

FranticRain, 2018-04-02:
https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-bluetooth

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain